lentinj/tp-compact-keyboard #18: Request: Swap FN and Ctrl buttons

MaZZly, 2015-01-13:
If possible =)

Will probably have gotten used to it by then but going back and forth between different layouts might mess wiht my head

Cheers

lentinj, 2015-01-13:
It's not I'm afraid. Or at least as far as I know.

Until it's possible under Windows (which AFAIK it isn't), then there's nothing that can be done for Linux. Your best course of action is to nag Lenovo on their forums.

closed 2015-01-13 by lentinj

MaZZly, 2015-01-13:
Nah.. Will mostly only use it when I'm on the go anyway =)

mithro, 2015-01-13:
Been a while since I looked at the code.

Is it possible to bubble the fn key up as a normal key event? Then Linux
can do it in the normal stack.

lentinj, 2015-01-13:
@mithro Nice idea, but don't think that'll work.

You do get a report when the Fn key is pressed, but it's the same report as you'd get for "all keys released" (note that holding down "f" then pressing Fn stops f being pressed). So it'd be hard to conclusively tell that Fn was pressed.

You don't get any report for key release. You can get Linux to fake these, but then the Fn would be immediately released, and you'd have no way to press another key whilst Fn is held down.

You can try this yourselves with the following as root:
```
mount -t debugfs /debug /debug
cat $(ls -d1 /debug/hid/000*\:17EF\:604* | tail -1)/events
```