lentinj/tp-compact-keyboard #29: Scrolling issues

K0ertis, 2015-10-29:
Hi! After upgrading to 15.10 I can not scroll any more. The middle button work fine but scrolling function is disabled. When I look at keyboard output on xev i get:
```
 LeaveNotify event, serial 37, synthetic NO, window 0x6000001,
    root 0xd8, subw 0x0, time 3801240, (99,101), root:(152,508),
    mode NotifyGrab, detail NotifyAncestor, same_screen YES,
    focus NO, state 16

 EnterNotify event, serial 37, synthetic NO, window 0x6000001,
    root 0xd8, subw 0x0, time 3801242, (99,101), root:(152,508),
    mode NotifyUngrab, detail NotifyAncestor, same_screen YES,
    focus NO, state 16

 KeymapNotify event, serial 37, synthetic NO, window 0x0,
    keys:  4294967256 0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    
 
 ButtonPress event, serial 37, synthetic NO, window 0x6000001,
     root 0xd8, subw 0x0, time 3801240, (99,101), root:(152,508),
     state 0x10, button 25, same_screen YES
 
 ButtonRelease event, serial 37, synthetic NO, window 0x6000001,
     root 0xd8, subw 0x0, time 3801240, (99,101), root:(152,508),
     state 0x10, button 25, same_screen YES
```
Not scrolling drives me nuts in coding... 
The tp-compact patches does not work for this issue. Btw. I changed the pointer by xmodmap to:
```
xmodmap -e "pointer = 1 25 3 4 5 6 7 8 9"
```
for disabled paste on middle click.

Thanks for any help

lentinj, 2015-10-29:
What kernel version are you using (``uname -a``)? Can you do a similar test with ``evtest`` outside Xorg?

K0ertis, 2015-10-29:
$ uname -a
```
Linux user-E5540 4.2.0-16-generic #19-Ubuntu SMP Thu Oct 8 15:35:06 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
I did: middlebutton press, scroll vertical and scroll horizontal.
evtest:
```
Event: time 1446114563.271593, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.271593, -------------- SYN_REPORT ------------
Event: time 1446114563.307577, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.307577, -------------- SYN_REPORT ------------
Event: time 1446114563.343576, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.343576, -------------- SYN_REPORT ------------
Event: time 1446114563.379587, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.379587, -------------- SYN_REPORT ------------
Event: time 1446114563.415548, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.415548, -------------- SYN_REPORT ------------
Event: time 1446114563.451547, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.451547, -------------- SYN_REPORT ------------
Event: time 1446114563.487577, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.487577, -------------- SYN_REPORT ------------
Event: time 1446114563.523546, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.523546, -------------- SYN_REPORT ------------
Event: time 1446114563.559578, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.559578, -------------- SYN_REPORT ------------
Event: time 1446114563.595583, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.595583, -------------- SYN_REPORT ------------
Event: time 1446114563.631554, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.631554, -------------- SYN_REPORT ------------
Event: time 1446114563.667587, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.667587, -------------- SYN_REPORT ------------
Event: time 1446114563.703546, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.703546, -------------- SYN_REPORT ------------
Event: time 1446114563.739543, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.739543, -------------- SYN_REPORT ------------
Event: time 1446114563.775577, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.775577, -------------- SYN_REPORT ------------
Event: time 1446114563.811578, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.811578, -------------- SYN_REPORT ------------
Event: time 1446114563.847575, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.847575, -------------- SYN_REPORT ------------
Event: time 1446114563.883541, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114563.883541, -------------- SYN_REPORT ------------
Event: time 1446114563.907120, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff0000fb
Event: time 1446114563.907120, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 0
Event: time 1446114563.907120, -------------- SYN_REPORT ------------
Event: time 1446114565.025958, type 4 (EV_MSC), code 4 (MSC_SCAN), value ff0000fb
Event: time 1446114565.025958, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 1
Event: time 1446114565.025958, -------------- SYN_REPORT ------------
Event: time 1446114565.275582, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.275582, -------------- SYN_REPORT ------------
Event: time 1446114565.311577, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.311577, -------------- SYN_REPORT ------------
Event: time 1446114565.347544, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.347544, -------------- SYN_REPORT ------------
Event: time 1446114565.383584, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.383584, -------------- SYN_REPORT ------------
Event: time 1446114565.419548, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.419548, -------------- SYN_REPORT ------------
Event: time 1446114565.455550, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.455550, -------------- SYN_REPORT ------------
Event: time 1446114565.491545, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.491545, -------------- SYN_REPORT ------------
Event: time 1446114565.527576, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.527576, -------------- SYN_REPORT ------------
Event: time 1446114565.563542, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.563542, -------------- SYN_REPORT ------------
Event: time 1446114565.599560, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114565.599560, -------------- SYN_REPORT ------------
Event: time 1446114565.635584, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
[... 384 lines cut ...]
Event: time 1446114568.443576, -------------- SYN_REPORT ------------
Event: time 1446114568.452166, type 2 (EV_REL), code 6 (REL_HWHEEL), value 2
Event: time 1446114568.452166, -------------- SYN_REPORT ------------
Event: time 1446114568.479556, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114568.479556, -------------- SYN_REPORT ------------
Event: time 1446114568.515538, type 1 (EV_KEY), code 274 (BTN_MIDDLE), value 2
Event: time 1446114568.515538, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1446114570.808667, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1446114570.808667, -------------- SYN_REPORT ------------
Event: time 1446114571.059574, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 2
Event: time 1446114571.059574, -------------- SYN_REPORT ------------
Event: time 1446114571.092441, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1446114571.092441, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1446114571.092441, -------------- SYN_REPORT ------------
```

lentinj, 2015-10-29:
Okay, I think this is the same as #23, compounded by the fact that you have wheel emulation turned on for the keyboard. You need to either:-
* Raise a bug with Ubuntu to backport these patches, or generate your own kernel with these patches (they went into upstream in Linux 4.3 https://lkml.org/lkml/2015/8/11/742 ):-
  * https://raw.githubusercontent.com/lentinj/tp-compact-keyboard/master/kernel-patch/3001-HID-lenovo-Use-constants-for-axes-names.patch
  * https://raw.githubusercontent.com/lentinj/tp-compact-keyboard/master/kernel-patch/3002-HID-lenovo-Add-missing-return-value-check.patch
  * https://raw.githubusercontent.com/lentinj/tp-compact-keyboard/master/kernel-patch/3003-HID-lenovo-Hide-middle-button-press-until-release.patch
* Turn wheel emulation off for your keyboard---if you just do this you will still need to filter your middle-button, but scroll events will return. It shouldn't be on in this case anyway, the keyboard does it itself.

K0ertis, 2015-10-30:
Hi! Last night my battery run out and I do not know if it correlates but today the scrolling is working again.

lentinj, 2015-10-30:
I'm guessing you enabled wheel emulation at some point with xinput, rebooting turned this back off again, so the hardware scroll events are allowed through again.

If you don't want middle-button presses, you're done. If you want to restore middle-button presses, you'll need the patches above / a 4.3 kernel.

closed 2015-10-30 by lentinj