lentinj/tp-compact-keyboard #31: A lenovo usb keyboard with a new HID_NAME

hrimfaxi, 2015-12-10:
Hi, I found out this utility even works with lenovo USB keyboard LXH-JME2209U. My Fn lock is successfully unlocked! Nice work!
However, its HID_NAME was "Lenovo Lenovo USB Keyboard". Maybe you could handle it too in the script.

# cat /sys/class/hidraw/hidraw2/device/uevent
DRIVER=hid-generic
HID_ID=0003:000017EF:00006018
HID_NAME=Lenovo Lenovo USB Keyboard
HID_PHYS=usb-0000:00:14.0-8/input0
HID_UNIQ=
MODALIAS=hid:b0003g0001v000017EFp00006018

dmesg:
[    1.144315] hidraw: raw HID events driver (C) Jiri Kosina
[    1.147175] usbcore: registered new interface driver usbhid
[    1.147177] usbhid: USB HID core driver
[    1.200747] hid-generic 0003:046D:C534.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-6/input0
[    1.260742] hid-generic 0003:046D:C534.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-6/input1
[    1.490841] hid-generic 0003:17EF:6018.0003: input,hidraw2: USB HID v1.11 Keyboard [Lenovo Lenovo USB Keyboard] on usb-0000:00:14.0-8/input0
[    1.570885] hid-generic 0003:17EF:6018.0004: input,hidraw3: USB HID v1.11 Device [Lenovo Lenovo USB Keyboard] on usb-0000:00:14.0-8/input1

lsusb -v:
Bus 001 Device 003: ID 17ef:6018 Lenovo 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x17ef Lenovo
  idProduct          0x6018 
  bcdDevice            1.40
  iManufacturer           1 Lenovo
  iProduct                2 Lenovo USB Keyboard
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     174
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0005  1x 5 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)

hrimfaxi, 2015-12-10:
Addition: seems only tp-compact-usb-keyboard.c works.

lentinj, 2015-12-10:
Okay, makes sense that only tp-compact-usb-keyboard works, given it's the same with the Lenovo USB keyboard I have. You could copy the top of the script into a new script that wraps around ``tp-compact-usb-keyboard.c`` if you wanted the auto-keyboard-finding functionality. Ideally we'd poke support for the keyboard into the Linux kernel though.

What hidraw device are you sending commands to, hidraw2 or hidraw3 (assuming you haven't unplugged anything since the above dmesg)? I'm guessing (well, more hoping) that it's the latter device, hidraw3, that works. Do all the special Fn-Key functions work properly if you use ``evtest`` on the linux console (i.e. outside X11)? Some might only be available on the second input device for the keyboard.

Does the keyboard have a pointing device of some kind too? It's hard to find out any information on it, best I can do is guess that it's the keyboard pictured here: http://www.newegg.com/Product/Product.aspx?Item=N82E16883108568

hrimfaxi, 2015-12-11:
Yeah I already done that part for udev to call tp-compact-usb-keyboard.

Either hidraw2 or hidraw3 works if I unplugged then plugged it back. It works even only one single hidraw device was given to tp-compact-usb-keyboard.

This keyboard have no pointing device. It was the keyboard shipped with Lenovo server and PC sold in China.
And quickly became a nightmare for people working on Linux with i+t. 

Here is the result of  xev | grep --line-buffered -A6 KeyRelease after running tp-compact-usb-keyboard /dev/hidraw3:
I think from F1 to F12 all keys are unlocked.
````
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77427322, (-131,235), root:(740,714),
    state 0x0, keycode 67 (keysym 0xffbe, F1), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77427978, (-131,235), root:(740,714),
    state 0x0, keycode 67 (keysym 0xffbe, F1), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77428490, (-131,235), root:(740,714),
    state 0x0, keycode 67 (keysym 0xffbe, F1), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77429010, (-131,235), root:(740,714),
    state 0x0, keycode 68 (keysym 0xffbf, F2), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77429690, (-131,235), root:(740,714),
    state 0x0, keycode 69 (keysym 0xffc0, F3), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77430754, (-131,235), root:(740,714),
    state 0x0, keycode 70 (keysym 0xffc1, F4), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77431314, (-131,235), root:(740,714),
    state 0x0, keycode 71 (keysym 0xffc2, F5), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77432098, (-131,235), root:(740,714),
    state 0x0, keycode 72 (keysym 0xffc3, F6), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77432674, (-131,235), root:(740,714),
    state 0x0, keycode 73 (keysym 0xffc4, F7), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77433587, (-131,235), root:(740,714),
    state 0x0, keycode 74 (keysym 0xffc5, F8), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77434186, (-131,235), root:(740,714),
    state 0x0, keycode 75 (keysym 0xffc6, F9), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77434874, (-131,235), root:(740,714),
    state 0x0, keycode 76 (keysym 0xffc7, F10), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77436218, (-131,235), root:(740,714),
    state 0x0, keycode 95 (keysym 0xffc8, F11), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 37, synthetic NO, window 0x2200001,
--
KeyRelease event, serial 37, synthetic NO, window 0x2200001,
    root 0xd4, subw 0x0, time 77437002, (-131,235), root:(740,714),
    state 0x0, keycode 96 (keysym 0xffc9, F12), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```