lentinj/tp-compact-keyboard #33: Can it work on mac os?

daimon99, 2016-03-13:
Hello, 

It is a great program.
I am using an iMac. Can it work on imac?
or could you help me to get it work?
thank you!

lentinj, 2016-03-14:
No, it uses the Linux hidraw devices which AFAIK aren't available on OS X.

Looking at this: http://stackoverflow.com/questions/6477385/raw-access-to-hid-devices-in-os-x, a rewrite using hidapi would work: http://www.signal11.us/oss/hidapi/

Looks like it'd be reasonably easy, but not something I'd have interest in.

unknownzerx, 2016-07-19:
I am using
- https://github.com/unknownzerx/tpkb
- https://github.com/unknownzerx/tpkb/releases/download/v1.0/tpkb.zip

lentinj, 2016-07-19:
Nice. Added a link to into the README.md here.

If you fancied supporting the USB version have a look at:-

https://github.com/lentinj/tp-compact-keyboard/blob/master/tp-compact-usb-keyboard/tp-compact-usb-keyboard.c

The key differences between supporting USB and bluetooth is the first byte of any command and using ``hid_send_feature_report()`` instead of ``hid_write()``.

closed 2016-07-19 by lentinj