lentinj/tp-compact-keyboard #35: Sometimes the touchpoint becomes "slow / laggy"

mithro, 2016-04-12:
I don't really know how to describe this otherwise. Occasionally the touch point becomes really slow to control the mouse pointer, it is kind of like pushing it through molasses. The mouse cursor does follow the path that would be expected if receiving all the input data (IE if you do a little circle motion, the cursor will follow the circle) but it takes a long time. Holding the power on the side of the keyboard until it turns off and then turning it back on again causes everything to become fine again.

This is with a Lenovo KT-1255 (Rev: W01) bluetooth keyboard.

Currently using the stock kernel driver with `4.4.0-13-generic #29~14.04.1-Ubuntu SMP Fri Mar 11 20:25:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux` on Ubuntu 14.04 (trusty). The kernel comes from these packages;
```
ii  linux-headers-4.4.0-13                                4.4.0-13.29~14.04.1                                 all          Header files related to Linux kernel version 4.4.0
ii  linux-headers-4.4.0-13-generic                        4.4.0-13.29~14.04.1                                 amd64        Linux kernel headers for version 4.4.0 on 64 bit x86 SMP
ii  linux-image-4.4.0-13-generic                          4.4.0-13.29~14.04.1                                 amd64        Linux kernel image for version 4.4.0 on 64 bit x86 SMP
ii  linux-image-extra-4.4.0-13-generic                    4.4.0-13.29~14.04.1                                 amd64        Linux kernel extra modules for version 4.4.0 on 64 bit x86 SMP
ii  linux-image-generic-lts-xenial                        4.4.0.13.7                                          amd64        Generic Linux kernel image
```

The bluetooth packages I have installed are;
```
ii  blueman                                               1.23-git201403102151-1ubuntu1                       amd64        Graphical bluetooth manager
ii  bluetooth                                             4.101-0ubuntu13.1                                   all          Bluetooth support
ii  bluez                                                 4.101-0ubuntu13.1                                   amd64        Bluetooth tools and daemons
ii  bluez-alsa:amd64                                      4.101-0ubuntu13.1                                   amd64        Bluetooth ALSA support
ii  bluez-cups                                            4.101-0ubuntu13.1                                   amd64        Bluetooth printer driver for CUPS
ii  bluez-gstreamer                                       4.101-0ubuntu13.1                                   amd64        Bluetooth GStreamer support
ii  bluez-hcidump                                         2.5-1                                               amd64        Analyses Bluetooth HCI packets
ii  bluez-tools                                           0.1.38+git662e-3                                    amd64        Set of tools to manage Bluetooth devices for linux
ii  bluez-utils                                           4.101-0ubuntu13.1                                   all          Transitional package
ii  gir1.2-gnomebluetooth-1.0                             3.8.2.1-0ubuntu4.2                                  amd64        Introspection data for GnomeBluetooth
ii  gnome-bluetooth                                       3.8.2.1-0ubuntu4.2                                  amd64        GNOME Bluetooth tools
ii  indicator-bluetooth                                   0.0.6+14.04.20140207-0ubuntu2                       amd64        System bluetooth indicator.
ii  libbluetooth-dev                                      4.101-0ubuntu13.1                                   amd64        Development files for using the BlueZ Linux Bluetooth library
ii  libbluetooth3:amd64                                   4.101-0ubuntu13.1                                   amd64        Library to use the BlueZ Linux Bluetooth stack
ii  libbluray1:amd64                                      1:0.5.0-1                                           amd64        Blu-ray disc playback support library (shared library)
ii  libgnome-bluetooth11                                  3.8.2.1-0ubuntu4.2                                  amd64        GNOME Bluetooth tools - support library
ii  liblua5.1-0:amd64                                     5.1.5-5ubuntu0.1                                    amd64        Shared library for the Lua interpreter version 5.1
ii  liblua5.2-0:amd64                                     5.2.3-1                                             amd64        Shared library for the Lua interpreter version 5.2
ii  liblua5.2-dev:amd64                                   5.2.3-1                                             amd64        Development files for the Lua language version 5.2
ii  obno                                                  11                                                  all          Out of the blue notifications.
ii  pulseaudio-module-bluetooth                           1:4.0-0ubuntu11.1                                   amd64        Bluetooth module for PulseAudio sound server
```

The output I have in dmseg is;
```
[2195004.182569] lenovo 0005:17EF:6048.004B: unknown main item tag 0x0
[2195004.183112] input: ThinkPad Compact Bluetooth Keyboard with TrackPoint as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/bluetooth/hci0/hci0:11/0005:17EF:6048.004B/input/input88
[2195004.183954] lenovo 0005:17EF:6048.004B: input,hidraw2: BLUETOOTH HID v3.09 Keyboard [ThinkPad Compact Bluetooth Keyboard with TrackPoint] on 5c:f3:70:66:70:71
[2196402.536747] init: tty1 main process ended, respawning
[2197655.882637] lenovo 0005:17EF:6048.004C: unknown main item tag 0x0
[2197655.882935] input: ThinkPad Compact Bluetooth Keyboard with TrackPoint as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/bluetooth/hci0/hci0:11/0005:17EF:6048.004C/input/input89
[2197655.883511] lenovo 0005:17EF:6048.004C: input,hidraw2: BLUETOOTH HID v3.09 Keyboard [ThinkPad Compact Bluetooth Keyboard with TrackPoint] on 5c:f3:70:66:70:71
```

The modules currently loaded are;
```
hid_lenovo             20480  0 
hid                   118784  4 hidp,hid_lenovo,hid_logitech,usbhid
```

lentinj, 2016-04-12:
I know what you mean. Although I'm not sure what can be done about it, if anything.

I'm reasonably sure (although got no links to prove it) that this is an issue under windows too, and is an artefact of Bluetooth going out of range. If you keep moving the cursor and take your keyboard further and further away, it'll eventually just drift in the last seen direction until you come back in range again.

I used to have this quite bad, until I noticed I'd left a (2.4Ghz) mobile hotspot between my keyboard and laptop. Moved it, solved problem. Of course there may be problems in bluez that means it struggles to recover from a poor connection, but I don't think it's particularly specific to these keyboards.

mithro, 2016-04-18:
Is there a way to get some type of error diagnostics?

lentinj, 2016-04-18:
Probably, but nothing I know of 'atop my head.