lentinj/tp-compact-keyboard #36: Successful pairing but trackpoint and mouse buttons input not working

nkoester, 2016-05-09:
Hi,

though its not directly related to the provided script, I hope to get some help on this here. I also have a lenovo comact bloutooth keyboard (KT-1255) which I intend to use on my arch linux box (Linux cheddar 4.5.1-1-ARCH #1 SMP PREEMPT Thu Apr 14 19:19:32 CEST 2016 x86_64 GNU/Linux). 

Pairing is possible, however I only can make key input. Trackpoint movement AND mouse buttons do not work. The signal seems to never be registered at all. Any idea? Any help on this?

Thanks in advance!
Cheers

lentinj, 2016-05-12:
Given the keyboard works I doubt there's anything wrong at the kernel level. It's far more likely to be a generic Xorg configuration issue, which you should take up in the arch forums. To help them debug, try:

* Use ``evtest`` outside Xorg on the console, do you see events for both keyboard/mouse?
* Use ``xev`` inside Xorg, do you see events then?

nkoester, 2016-05-12:
Okay thanks for the pointers. For future reference: Unfortunately both returned nothing for the mouse or trackpoint (hoverver all key inputs return useful events). The keybord DID work once, I will check it on some other systems before I will ask in the arch forums. Thanks again.

bmayton, 2016-05-19:
I recently started having the exact same problem, and found this issue when searching for it.

The keyboard worked well for a while paired with a machine running Arch.  The initial behavior when the trackpoint stopped working was that the cursor would move up and to the left with the keyboard connected, and moving the stick had no effect.  I don't recall testing if the trackpoint buttons worked at the time.  After another couple of connection attempts, which may have included re-pairing the keyboard, the trackpoint stopped working entirely, including the buttons.  The keyboard still works fine.  The behavior is exactly the same when paired with a Windows machine with the official Lenovo drivers.

No events (evtest, xev) or Bluetooth messages (hcitool) are generated when moving the TrackPoint stick or pressing the mouse buttons.  The fact that the buttons stopped working leads me to believe that the firmware has completely disabled pointing device functionality—the mouse buttons are wired as part of the keyboard matrix, but don't work while the keyboard still functions perfectly.  The strain gauges in the TrackPoint are wired to the mainboard via a separate 4-pin connector, and the resistances seem reasonable (and change when the stick is deflected).

I wonder if there is an undocumented function for disabling the TrackPoint that somehow got activated, or if the internal configuration has somehow become corrupted.  Forcing the device to un-pair (Fn+Delete) and re-pairing doesn't change anything, nor does powering down the keyboard with the switch.  Disconnecting the battery for a while also doesn't seem to reset anything (the keyboard retains its pairing).

There are UART test points on the bottom of the main board (probably how the firmware is initially programmed; they had probe marks on them) but there is no activity on the transmit line when the keyboard powers on.

I basically gave up and bought another one (it's unfortunately the only device that meets my requirements—wireless with a pointing device with a physical middle button).  I'm still poking at the broken one though—it would be great to bring it back to life.

bmayton, 2016-06-01:
Following up on this as I managed to repair the broken TrackPoint.  Unfortunately, it is a hardware issue and requires reflow soldering equipment to fix.

Most ThinkPad keyboards I've seen put the controller for the TrackPoint on a small board attached to the TrackPoint itself that reads the strain gages and mouse buttons and provides PS/2 output.  On the Bluetooth keyboard, the cable coming off of the TrackPoint connects directly to the strain gages; the controller is on the main board.  The main board in these keyboards has two large ICs; one is a Broadcom bluetooth SOC on a module mounted to the main PCB.  The other (U2) is a chip I haven't been able to identify, perhaps an ASIC as there is no programming header (marked 502A6 HF372 7AV1).  The keyboard matrix is wired directly to the Broadcom SOC.  The TrackPoint and mouse buttons go to U2.  (In my previous comment I stated that the mouse buttons are part of the keyboard matrix; while they share the same connector they are not actually wired to the same place).

U2 thus appears to just be the controller for the TrackPoint, which communicates with the SOC over I²C instead of PS/2.

Inspection under a microscope revealed poor soldering on U2.  Removing the chip did not change the behaivor (keyboard still paired and keyboard worked, but no pointing device or buttons).  Carefully cleaning the pads and the chip (some of the pins were oxidized and it didn't look like they'd ever had a good solder connection) and re-soldering it fixed the problem—the TrackPoint works once again.  The fix is therefore straightforward with reflow soldering equipment and experience; however as U2 is a QFN package, sending the keyboard in for service (or accepting the fact that it's now keyboard-only) is probably the only option without reflow equipment.

lentinj, 2016-06-01:
@bmayton Thanks for the update. Whilst my own bluetooth keyboards seem fine, my USB keyboard drifts so much nowadays it's pretty much unusable. Way more than could be explained by calibration configuration. A similar QA problem could explain matters, but unfortunately my best reflow equipment is a pie dish and a hot air gun.

I guess this also means the keyboard isn't a laptop spare part with an extra circuit board, which is a shame. Out of interest, did you make a note / take photos of what the Broadcom SoC is?

nkoester, 2016-06-02:
@bmayton Wow, thanks for all this information! I actually made use of my warranty and Lenovo sent me a new keyboard which is working just fine with my setup. Luckily, they do not want the old one back! This would allow me to verify your work and repair my keyboard as well :). 

Unfortunately I also do not have the according equipment and soldering skills to perform this task. Do you think that - in any way - a normal manual soldering would work on this (this is what I have direct access to)? Further, is it possible for you to make some pictures and give me some further info on what to solder where? I think this would be very useful for other who find this info and maybe I can find someone with the according skills and equipment to help me...Thanks in advance!

lentinj, 2016-06-02:
> Do you think that - in any way - a normal manual soldering would work on this

No, have a look at the picture on https://en.wikipedia.org/wiki/Quad_Flat_No-leads_package - where would you put the soldering iron? My pie tin and hot air gun reference wasn't entirely facetious, Google that for methods or if your feeling brave, or don't care since you've got a replacement.

bmayton, 2016-06-05:
The Broadcom chip is a BCM20730.

Repair with a good soldering iron and a good flux is not entirely out of the question.  I do touch-up work on 0.5mm QFNs with an iron fairly frequently.  Since the pads are too small to be soldered individually, it requires some trust in the flux and the surface tension of the solder, but it definitely can be done.  I don't have the right camera with me to shoot video through the microscope right now, but I could shoot a quick example next week if there was interest.

I've put some more information and pictures about repairing the keyboard here:
http://bdm.cc/2016/06/lenovo-bluetooth-keyboard-repairs/

lentinj, 2016-06-06:
> I've put some more information and pictures about repairing the keyboard here:
> http://bdm.cc/2016/06/lenovo-bluetooth-keyboard-repairs/

Awesome write-up, thanks :) I've linked to it in the README in this project, in case it's useful for others.

nkoester, 2016-06-17:
Well, it took me some time but I finally managed to give it a go and follow @bmayton s awesome write-up (thanks so much!) in order to repair the trackpoint. Unfortunately, I was not successful in my try and the trackpoint is still not responding. I blame my bad equipment (aka. my giant iron tip) as well as my very poor soldering skills ;) Further I had no microscope or magnifying glass so I pretty much tried to do it "blind"...

But still, I do not give up on this keyboard! There is proper equipment at my work and I will give it another go there. Maybe also someone with better skills there can help me out! I would love to be able to repair this thing - even though I do have a replacement.