lentinj/tp-compact-keyboard #37: Tap to click?

SohumB, 2016-05-12:
Hey! 

I've been trying to enable tap-to-click on one of these trackpoint compact keyboards (usb), but all the help I've seen involves paths under `/sys/devices/platform` that I don't have. Any idea of what would be involved in making this happen?

Thanks!

lorenzhs, 2016-05-12:
Uh, there exists models with a touchpad? :open_mouth:

I think you need to do this in your xorg / libinput configuration, it shouldn't have anything to do with the driver

SohumB, 2016-05-12:
Sorry! I meant press-to-click, of the trackpoint. Is that still unrelated to the driver?

lentinj, 2016-05-12:
Yeah, any paths there aren't going to exist, firstly because they're connected via. USB rather than an approximation to PS/2 inside your laptop. Secondly because there's no direct access to the trackpoint on these keyboards.

As with Fn<->Ctrl swapping, the way forward is to find out if this is possible under Windows first, then we can sniff the connection to see what is going on. Last time I tried the windows software I couldn't see an option for this, so this is likely impossible. It might be worth asking Lenovo support / forums, and see if you get a response?

( @lorenzhs I think what they mean is pressing the trackpoint in to select, they're pressure sensitive. I remember it being an option back in the Windows 2000 trackpoint driver)

lorenzhs, 2016-05-12:
Ah, I see. But I found an external thinkpad keyboard with a touchpad, it's called the "ThinkPad UltraNav keyboard", apparently that does exist ;) [image]

lentinj, 2016-05-12:
Yeah, there's one without the numeric keypad too I think. And (keeping it on topic) for that generation keyboard press-to-select will be an option, as you get direct access to the trackpoint. Or at least I think so, I don't actually own one.

However, they were supposed to be quite spongey and bendy in comparison to typing on an actual laptop. The compact USB keyboards were the first that fixed this by gluing the keyboard to a thick chunk of aluminium (I know it's glued since I tried to take my USB one to bits...).

SohumB, 2016-05-12:
I actually do own one of those, but I've never tried to enable press-to-click on it. But, yes, I far prefer the typing experience on the compact one!

Okay, thank you both for the help! I'll go poke lenovo support, next :)

closed 2016-05-12 by SohumB