lentinj/tp-compact-keyboard #38: usb keyboard sends fn key to event1.

amosbird, 2016-06-21:
I'm using compact usb keyboard on Archlinux. Using `evtest` I get this two keyboard inputs, 
```
/dev/input/event0:	Lenovo ThinkPad Compact USB Keyboard with TrackPoint
/dev/input/event1:	Lenovo ThinkPad Compact USB Keyboard with TrackPoint
```
All the normal keys inputs are from event0 while trackpoint events come through event1. But fn functional keys are routed to event1 which leads to irresponsibility of these events by the X server.  xev shows empty keycodes for `F1`, `F2`, `F3`, and gets nothing from `F4`,
```
KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  4294967265 0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyGrab, detail NotifyAncestor

FocusOut event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyUngrab, detail NotifyPointer

FocusIn event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyGrab, detail NotifyAncestor

FocusOut event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyUngrab, detail NotifyPointer

FocusIn event, serial 34, synthetic NO, window 0x2a00001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 34, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0 
```

Is there a way to fix this?

lentinj, 2016-06-21:
> All the normal keys inputs are from event0 while trackpoint events come through event1. But fn functional keys are routed to event1

It's not entirely obvious why, but this seems to be the way the keyboard works. It'd be hard work to fix, and I'm not sure what you'd gain.

> But fn functional keys are routed to event1 which leads to irresponsibility of these events by the X server.

This sounds like fn_lock has been disabled, and pressing F1 is being interpreted as a mute key by your window manager. What does ``cat /sys/bus/hid/devices/*17EF\:604*/fn_lock`` return?

amosbird, 2016-06-21:
It returns `1`.  My keyboard's model is KU-1255.

closed 2017-05-17 by amosbird