lentinj/tp-compact-keyboard #40: README updates for bluetoothctl setup

c-j-s, 2016-10-17:
I successfully set up my keyboard (a KT-1255, Japanese layout) using Arch Wiki's generic [Bluetooth](https://wiki.archlinux.org/index.php/Bluetooth) page (rather than the [Bluetooth Keyboard] (https://wiki.archlinux.org/index.php/Bluetooth_keyboard#Pairing_process) page mentioned in the `bluetoothctl` section of `README.md`).

Rather than using `hcidump` to get the passcode, I let `bluetoothctl` print it, but that did require the same fix or workaround mentioned in the  [Logitech keyboard does not pair](https://wiki.archlinux.org/index.php/Bluetooth#Logitech_keyboard_does_not_pair) troubleshooting section of that page, _viz._

    hciconfig hci0 sspmode 0

I'm not clear on how or if this relates to the `bluez-test-*` method of setting up the keyboard, but I do suspect that it would allow `bluetooth-wizard` to correctly display the passcode if it has that capability at all. Perhaps someone can test it and leave a note here?

I'm happy to update the README for either the `bluetoothctl` section or that and the `bluetooth-wizard` section and submit a pull request if that's helpful to you.

lentinj, 2016-10-17:
Thanks for letting me know!

> I'm not clear on how or if this relates to the ``bluez-test-*`` method of setting up the keyboard

The "somewhat quirky" paring method I describe in the README is actually [simple secure pairing](https://en.wikipedia.org/wiki/Bluetooth#Pairing_mechanisms), which back when ``bluez-test-*`` still existed wasn't supported by the userland tools, so you had to use ``hcidump`` to find out the code that bluez was treating as an unknown message and ignoring.

``hciconfig hci0 sspmode 0`` instead disables support for SSP entirely, which would have been a cleaner way of pairing instead of faffing with ``hcidump``. However, in my experience SSP works fine with a reasonably modern Linux distro and pairing should just work.

I should have some time this evening to rewrite this bit of the README, will let you know when it's done.

lentinj, 2016-10-17:
Re-arranged things, put the old pairing instructions to the bottom, mentioned your tip about using ``sspmode 0``. Feel free to raise a pull request if you think extra instructions would be useful.

closed 2016-10-17 by lentinj

c-j-s, 2016-10-18:
Looks good to me. Thanks!