lentinj/tp-compact-keyboard #42: Trackpoint sensitivity doesn't work well.

amosbird, 2017-04-01:
By setting sensitivity to 200, I get the needed speed but the point resolution is pretty slow. I can see the zig zag tragetery when moving trackpoint. Thinkpad builtin keyboard has sensitivity and speed settings. Should we add the speed setting for tp-compact-keyboard?

amosbird, 2017-04-02:
hmm, can we export a `speed` file node in the dev path?
```
 ❯ find /sys -name speed
...
/sys/devices/platform/i8042/serio1/serio2/speed
...

 ❯ find /sys -name sensitivity
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-7/2-7:1.0/bluetooth/hci0/hci0:256/0005:17EF:6048.0006/sensitivity
/sys/devices/platform/i8042/serio1/serio2/sensitivity
```

The built-in trackpoint driver has this and it works quite well.

lentinj, 2017-04-02:
No, as there's no way of setting the speed for the trackpoint. These aren't standard trackpoint devices (at least by the time the OS gets to see them), and a lot of things that apply for PS/2-interfaced trackpoints will not work here.

closed 2017-04-02 by lentinj

amosbird, 2017-04-02:
That's quite unfortunate. Can  it be solved by adjusting the driver?

lentinj, 2017-04-03:
Nope. Not unless the keyboard has features that Windows doesn't use (and thus nobody knows about).