lentinj/tp-compact-keyboard #43: Mutte mic button doen't work

rcdeoliveira, 2017-12-11:
I am running Debian Buster with kernel version 4.13.0 and mic off / on button doesn't work. How can enable it?

P.D:
Thanks for share this code, it's awesome

lentinj, 2017-12-11:
This isn't a problem specific to these keyboards I don't think, by default the Mic mute key doesn't do anything.

Have a look at: http://www.thinkwiki.org/wiki/Microphone_Mute_Button

rcdeoliveira, 2017-12-11:
I read that same page before create the issue, but my t440s keyboard works
just fine, even the on/off mic button.

mrPjer, 2017-12-13:
To confirm this - the mic mute button on my X220 works, but it doesn't work on the USB Compact Keyboard.

lentinj, 2017-12-15:
If it's working in one not the other, then you're probably experiencing a bug with the systemd keymaps. See the "F20" section at the bottom of that page. Your internal keyboard has probably got it's MICMUTE key remapped to F20, but the external keyboard is still mapped to MICMUTE.

Either report it with systemd or use hwdb to change your keymap locally.

mrPjer, 2018-01-13:
Using showkey, my internal mute button gives a scancode 0x5a 0xda and a keycode 190. The external keyboard gives a keycode 248, but no scancode, while all the other keys give a scancode.

Can I remap this without a scancode?

lentinj, 2018-01-19:
No, you can't have no scancode. It just means it's been filtered somewhere along the line. Heed the warnings in showkey's manpage, evtest is a better tool to use.