lentinj/tp-compact-keyboard #44: ThinkPad Multi Connect Bluetooth Keyboard with Trackpoint support

bmwcmw, 2018-03-16:
Hi,

I have a ThinkPad Multi Connect Bluetooth Keyboard with Trackpoint device.

By running directly `/bin/echo -ne "\x18\x05\x01" | dd of=/dev/$HIDRAW 2>/dev/null` it doesn't make any effect.

I also tried to look in `/sys/bus/hid/devices/0005\:17EF\:6083.0003/`                      
country            hidraw/            modalias           power_supply/      subsystem/         
driver/            input/             power/             report_descriptor  uevent       

there's nothing named fn_lock.

Any idea to make it works ? Thanks.

lentinj, 2018-03-19:
The device ID isn't one recognised by the kernel HID driver, which will be why there's no fn_lock file. And if your echo command doesn't work there's not much point adding it (assuming you set ``$HIDRAW`` first).

The closest known device ID in hid-ids.h is USB_DEVICE_ID_LENOVO_X1_COVER (0x6085). At a guess is this the Thinkpad Tablet 2 Bluetooth keyboard? e.g. https://www.aliexpress.com/item/New-original-Lenovo-ThinkPad-Tablet-2-Bluetooth-Keyboard-with-Stand-US-English-0B47270/32810910666.html

What does ``cat /sys/bus/hid/devices/0005:17EF:6083.0003/input/input*/name`` return?

lentinj, 2018-03-27:
Apologies, I thought it was a typo. I've never head of the *Multi Connect* (not Compact) keyboard until now. My first thought was it's a fake, but it does seem to be a thing (KT-1525 FRU 03X7111):

* http://webdoc.lenovo.com.cn/lenovowsi/new_cskb/att/147372/%E5%BF%AB%E9%80%9F%E4%BD%BF%E7%94%A8%E6%8C%87%E5%8D%97.pdf
* http://think.lenovo.com.cn/htmls/knowledge/detail_20150826174339180.html
* http://jonaschen.blogspot.co.uk/2018/01/thinkpad-bluetooth-keyboard.html

Did the keyboard come with any drivers for any OS? The ranting here http://bbs.thinkworldshop.com.cn/thread-1767307-1-1.html suggests that there might not be any publicly available.

lentinj, 2018-03-28:
However, back to your problem. Given it tries to support Windows/iOS/MacOS I'd be surprised if it relied on custom support in the OS to work. There's no drivers shown in http://webdoc.lenovo.com.cn/lenovowsi/new_cskb/att/147372/%E5%BF%AB%E9%80%9F%E4%BD%BF%E7%94%A8%E6%8C%87%E5%8D%97.pdf either. I notice that the keyboard doesn't have "FnLk" written on it anywhere, unlike the Compact keyboard this repository supports. It may not have a "FnLk" mode at all.

What happens if you run ``evtest`` and press your FN keys? Do you get different ``MSC_SCAN`` values for F1 and Fn-F1? What do you get if you press and release Fn on it's own?

With luck you can get away with writing a udev ``hwdb`` file for the keyboard that swaps around the F1 and Fn-F1 scancodes.

bmwcmw, 2018-03-29:
Ha yes KT-1525 FRU 03X7111 is the right model written on the back of keyboard. 

It comes with only one paper in the box(your pdf link) and without any driver found. Indeed it's not bad for Android/Linux devices except the fn lock, but in Windows the "middle key + red dot" scrolling doesn't work as usual.

When I run evtest, with keys pressed
- Fn
  type 4 (EV_MSC), code 4 (MSC_SCAN), value 70072
  type 1 (EV_KEY), code 193 (KEY_F23), value 1

- F1(Mute)
  type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
  type 1 (EV_KEY), code 113 (KEY_MUTE), value 1

- Fn + F1 
  type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003a
  type 1 (EV_KEY), code 59 (KEY_F1), value 1

The events are different. 

I'll try to override hwdb settings using it's modalias "hid:b0005g0001v000017EFp00006083" or the name, thanks for your advice.

lentinj, 2018-04-05:
> in Windows the "middle key + red dot" scrolling doesn't work

Presumably under Linux the general middle-button-scrolling options do the trick here?

I don't know of any facilities in the kernel to manage an Fn Key (like it does for num-lock or caps-lock), it was something I considered at one point with the Compact Keyboards. There may be something hidden within Xorg keyboard mappings that could manage it though.

lentinj, 2018-04-05:
The note in the README is probably all I can do for this. If you've got any useful examples I'll happily add them / link to them. I don't think I can justify getting one myself.

closed 2018-04-05 by lentinj