lentinj/tp-compact-keyboard #46: automatically enable Fn-Lock

sonnyp, 2019-07-11:
I have the following udev rule:

/etc/udev/rules.d/10-thinkpad-usb-keyboard.rules 
```
SUBSYSTEM=="usb", ATTRS{idVendor}=="17ef", ATTRS{idProduct}=="6047", RUN+="/opt/thinkpad-usb-keyboard-hotkeys"
```
and script:

```sh
#!/bin/sh
FILENAMES=(/sys/bus/hid/devices/*:17EF:6047*/fn_lock)

for FILENAME in ${FILENAMES}
do
  echo 0 > ${FILENAME}
done
```

It works well, but is there a better way?

lentinj, 2019-07-15:
Not really. You could probably simplify this by using udev variable substitutions to find the path, something like ``RUN += "/bin/sh -c 'echo 0 > $devpath/fn_lock'"``, but it's not a dramatic difference.

amosbird, 2019-10-16:
Hmm, but I cannot see any fn_lock files on arch

```
 ❯ ls /sys/bus/hid/devices/*
'/sys/bus/hid/devices/0003:17EF:306F.001F':
country  driver  hidraw  input  modalias  power  report_descriptor  subsystem  uevent

'/sys/bus/hid/devices/0003:17EF:3077.001E':
country  driver  hidraw  modalias  power  report_descriptor  subsystem  uevent

'/sys/bus/hid/devices/0003:17EF:60B5.002C':
country  driver  hidraw  input  modalias  power  report_descriptor  subsystem  uevent

'/sys/bus/hid/devices/0003:17EF:60B5.002D':
country  driver  hidraw  input  modalias  power  report_descriptor  subsystem  uevent

'/sys/bus/hid/devices/0003:17EF:60B5.002E':
country  driver  hidraw  input  modalias  power  quirks  report_descriptor  subsystem  uevent

```

sonnyp, 2019-10-16:
@amosbird you don't see to have any Thinkpad Compact keyboard

```sh
ls /sys/bus/hid/devices/*:17EF:6047*/
```

amosbird, 2019-10-16:
I don't. But I'm refering to this

> Linux kernels 3.17 and above (or any kernel with the kernel-patches applied) have built in support for the keyboards. This means all keys should work out the box, and you can control whether fn_lock is enabled by:

sonnyp, 2019-10-16:
it's referring to Thinkpad Compact keyboards

amosbird, 2019-10-17:
Ah, I see. But it's still a thinkpad keyboard, hacked from x1 tablet 3.

[image]

amosbird, 2019-10-17:
I'm trying to base on the hid-lenovo.c and add fn_lock support for x1-tab3. Is there any way of finding the hid events for that keyboard's fn_lock functionality?

lentinj, 2019-10-17:
The patches / utilities in here are for the compact keyboards, and will not support anything else. The ``fn_lock`` files are compact keyboard specific, and not how most Thinkpad keyboards work, where the Fn key is managed by the Embedded Controller.

> I'm trying to base on the hid-lenovo.c and add fn_lock support for x1-tab3

https://github.com/lentinj/tp-compact-keyboard#how-i-did-it

Regardless, this is definitely off-topic for this issue.

commit d3a76c91 (2019-10-18): README: Notes on disabling FnLk. Fixes #46
closed 2019-10-18 by lentinj