lentinj/tp-compact-keyboard #47: TP keyboard, no hid-lenovo (and no fn_lock)

Jellby, 2019-11-21:
Just installed a new Kubuntu 18.04 system, but the TP keyboard (just keyboard, no pointing device) does not seem to be recognized as such. In Kubuntu 14.04 I managed to get tp-compact-usb-keyboard working. Can I force the keyboard to use `hid-lenovo` somehow?

```
$ lsusb
Bus 001 Device 008: ID 17ef:6018 Lenovo

$ dmesg
[10072.863924] usb 1-11: new low-speed USB device number 9 using xhci_hcd
[10073.037136] usb 1-11: New USB device found, idVendor=17ef, idProduct=6018
[10073.037141] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10073.037145] usb 1-11: Product: Lenovo Low Profile USB Keyboard
[10073.037148] usb 1-11: Manufacturer: Lenovo
[10073.050469] input: Lenovo Lenovo Low Profile USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/0003:17EF:6018.000D/input/input28
[10073.108598] hid-generic 0003:17EF:6018.000D: input,hidraw1: USB HID v1.11 Keyboard [Lenovo Lenovo Low Profile USB Keyboard] on usb-0000:00:14.0-11/input0
[10073.125883] input: Lenovo Lenovo Low Profile USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.1/0003:17EF:6018.000E/input/input29
[10073.184532] hid-generic 0003:17EF:6018.000E: input,hidraw2: USB HID v1.11 Device [Lenovo Lenovo Low Profile USB Keyboard] on usb-0000:00:14.0-11/input1

$ udevadm info -a -p /sys/bus/hid/devices/0003:17EF:6018*
  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/0003:17EF:6018.000D':
    KERNEL=="0003:17EF:6018.000D"
    SUBSYSTEM=="hid"
    DRIVER=="hid-generic"
    ATTR{country}=="00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0':
    KERNELS=="1-11:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="usbhid"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="03"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="01"
    ATTRS{bNumEndpoints}=="01"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-11':
    KERNELS=="1-11"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="8"
    ATTRS{bMaxPower}=="100mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 2"
    ATTRS{bcdDevice}=="0220"
    ATTRS{bmAttributes}=="a0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="9"
    ATTRS{devpath}=="11"
    ATTRS{idProduct}=="6018"
    ATTRS{idVendor}=="17ef"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Lenovo"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="Lenovo Low Profile USB Keyboard"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="removable"
    ATTRS{speed}=="1.5"
    ATTRS{urbnum}=="416"
    ATTRS{version}==" 2.00"

```

lentinj, 2019-11-21:
It's not as easy as forcing binding with ``hid-lenovo`` unfortunately, since the driver won't know what to do with it. You'd need to add it's product ID in roughly the same places as ``USB_DEVICE_ID_LENOVO_CUSBKBD``, for example:

https://elixir.bootlin.com/linux/v5.3.12/source/drivers/hid/hid-lenovo.c#L837

Obviously for this to be a proper patch it would need to not add the various trackpoint related options, which can't be that useful on your keyboard.

Jellby, 2019-11-22:
I see... Well, `tp-compact-usb-keyboard.c` works fine anyway, and I think this udev rule works too:

    SUBSYSTEM=="hidraw", ATTRS{idVendor}=="17ef", ATTRS{idProduct}=="6018", RUN+="/path/to/tp-compact-usb-keyboard %E{DEVNAME}"

lentinj, 2022-09-28:
> multimedia keys (including the calculator key) apparently send no event (according to xev at least).

Have you tried with ``evtest`` to see what the kernel is seeing? If it's producing events, it's getting lost somewhere in X11/libinput (which seems reasonably likely).

> The keyboard looks like this, by the way

Yup, annoyingly I found one in the street and could have grabbed it for testing, but when I came back it had already gone. Will be more eager next time I find street-keyboards :)

Jellby, 2022-09-28:
Alright, `evtest` shows some stuff: `KEY_MUTE`, `KEY_VOLUME_DOWN`, etc. with `value 1` when pressed and `value 0` when released (on a different machine, so I can't copy-paste, sorry).

Jellby, 2022-09-28:
This looks like the relevant issue: https://gitlab.freedesktop.org/libinput/libinput/-/issues/745

I also see `Lenovo Lenovo Low Profile USB Keyboard Consumer Control: device is a joystick or a gamepad, ignoring`. That was fixed 6 months ago, I won't hold my breath for it to be updated in Ubuntu 22.04...