lentinj/tp-compact-keyboard #48: Thinkpad Multi Connect details

tinywrkb, 2019-12-19:
I grabbed one of those unicorns to use with a tablet and thought I might post here some details and my findings so they can be added to the device's readme.

TLDR: Quality device with zero support and in its current state not recommended for Windows and Android users. iOS users can disable

### Details:

* HID Name: ThinkPad Multi Connect Bluetooth Keyboard with Trackpoint
* Model: KT-1525
* P/N: 4X30K12182
* Fru P/N: 03X7111
* Vendor ID: 0x17ef
* Product ID: 0x6083
* MFG: CEM 2019-04-02

This is a quite nice keyboard quality wise, I also have the USB variant KT-1255 which feels cheap and flimsy while the Multi Connect is robust has a very solid shell while not weighting more (well it's about ~30 grams more), and the surface has a better finish with nicer texture and matte color.  
Also, the it has better switches which feel damper and giving more linear feedback while the key travel seem more or less the same.  
In contrast to my USB keyboard, the Multi Connect is stable on the desk even with just the rubber feet and I don't need use the foldable legs.  
As my USB keyboard manufactured in 2017 and the Multi Connect in 2019, it's possible this an update to all the Thinkpad keyboard models.
Overall, the keyboard feels and looks premium.

The keyboard supports changing the Trackpoint sensitivity with Fn+Up/Down Arrow.  
The paired device type can be change to one of the following: Windows, Android, and iOS with Fn+Delete. This seems to effect only some of the keyboard keys (like Super) and not the mouse buttons. I found it a useless feature, especially as it does not affect the middle button behavior while scrolling.  
The three Bluetooth keys replaced Home, End and Insert which is a little confusing, Fn+Right/Left Arrow adds Home & End back but the Insert key is nowhere to be found.  
There's a tiny toggle switch at the side of the keyboard for turning it on/off so it can be carried in a bag without worrying about turning it on by mistake and discharging the battery. 

This is a Bluetooth device only so while it doesn't appear as an USB keyboard when the cable is connect, it can be used while charging the battery.

### Issues

The usual Thinkpad keyboard driver for Windows is still not compatible and there doesn't seem to be any other driver.  
This creates a problem in which the middle button send key press&release event while scrolling with the Trackpoint.  
I quickly glanced at the lenovo-hid Linux driver and I'm guessing maybe this can be fixed for Linux but I'll don't believe I'll get to this in the near future.  
I also did not tried to read the exact HID codes that the keyboard sends during scrolling and pressing the middle button.  
I should mention that Baidu search didn't help (with the help of Google translate) nor the merchant I bought the keyboard in locating Windows drivers.

The second issue is the missing Fn-Lock feature and the fact the keyboard default to the media keys.  
I tried a brute approach to find if the current FW has such feature and failed. The only thing I found is that it accept writing everything the range (0x02, 0x01, 0x00-0xFF) and  (0xCC, 0x01, 0x00-0xFF), one time just the odd numbers (for the last of the triple) and other time just the even, but this doesn't seem to make any effect on the keyboard.  
I might have made the keyboard go haywire while trying to write to the other ranges but if not the FW might not offers the feature.
I guess that for Linux we can just reverse the key mappings in the kernel code.

I might have some time in a couple of weeks to figure out what can be improve in the kernel for better out of the box behavior with the keyboard.

lentinj, 2019-12-20:
> This is a quite nice keyboard quality wise, I also have the USB variant KT-1255 which feels cheap and flimsy while the Multi Connect is robust has a very solid shell while not weighting more (well it's about ~30 grams more), and the surface has a better finish with nicer texture and matte color.

I thought my USB KT-1255 was terrible, sounds like yours is even worse :) Externally it looks identical to the compact bluetooth ones I have (including rubber feet), just typing is terrible and the trackpoint permanently drifts.

> The usual Thinkpad keyboard driver for Windows is still not compatible and there doesn't seem to be any other driver.

The first time someone mentioned the existence of this keyboard I did rummage and the only thing I found was a PDF of the 1-page manual, which has no mention of any possible driver.

> I guess that for Linux we can just reverse the key mappings in the kernel code.

For the compact and tablet 2 keyboards I have, the default mode doesn't distinguish between roughly F7..F12, so even that isn't very useful. Of course this may be different, given it's expected to work.

If I find out the commands for the tablet2 keyboard I'll let you know, that at least has a driver. I just need to persuade it to install on a VM that isn't a tablet2.

Mind if I grab some of this text for https://github.com/lentinj/tp-compact-keyboard/tree/master/tp-multi-connect/README.md?

tinywrkb, 2019-12-20:
> Mind if I grab some of this text for https://github.com/lentinj/tp-compact-keyboard/tree/master/tp-multi-connect/README.md?

Updating that readme is one of the reasons I posted here so I don't mind at all.

> just typing is terrible and the trackpoint permanently drifts.

Typing om my KT-1255 does leave something to desire, to say the least, but I don't have trackpoint drifting problem.

> The first time someone mentioned the existence of this keyboard I did rummage and the only thing I found was a PDF of the 1-page manual, which has no mention of any possible driver.

The same single page manual and what seems like warranty details (didn't tried to translate the latter) were the only papers that came in the product packaging.  
Considering this isn't a single batch but manufactured for a couple of years (definitely 2017-2019), I pretty surprised a Windows driver is nowhere to be found. The middle click while scrolling and no Fn-Lock feature makes using the keyboard in Windows unbearable.

guanghwang, 2020-01-14:
Thanks for the information.
I owned this keyboard as well. I mainly use it under Linux (swaywm) and may provide some testing work if needed. Also, Chinese is my native language. If you need to baidu something, feel free to contact me.

Nightowl1lt, 2022-04-19:
I just bought this keyboard 2ndhanded today and the lack of driver is truly unbearable. Just want to have an update if you guys found any solution to ctr-fn switch and fnlock  function?

jeroney, 2022-04-19:
Really no way to switch ctr-fn.
Though you can disable fnlock via udev by default. 

While working on testing  the new thinkpad keyboard 2 (patches still making way into kernel) I noticed that bluetooth shows up differently in udev. So made something that works for both  keyboards.

If this helps. Here is bash script to write the rule, then reboot or restart udev.

cat <<'EOF' > /etc/udev/rules.d/99-disable-fn-lock-thinkpad-trackpoint-keyboards.rules

#ThinkPad TrackPoint Keyboard I & II USB
SUBSYSTEM=="hid", \
DRIVER=="lenovo", \
ATTRS{idVendor}=="17ef", ATTRS{idProduct}=="6047|60ee", \
ATTR{fn_lock}="0"
#ThinkPad TrackPoint Keyboard I & II Bluetooth
SUBSYSTEM=="input", \
ATTRS{id/vendor}=="17ef", ATTRS{id/product}=="6048|60e1", \
TEST=="/sys/$devpath/device/fn_lock", \
RUN+="/bin/sh -c 'echo 0 > \"/sys/$devpath/device/fn_lock\"'"
EOF

lentinj, 2022-04-19:
@jeroney The Thinkpad multi-connect is yet another keyboard, roughly a "trackpoint keyboard 1.5", it doesn't seem to respond to the commands the originals do (AIUI).

I've not seen anything new beyond what's in this issue, but haven't looked that hard. Maybe I should, given the trackpoint in my KT-1525 is dying.

jeroney, 2022-04-19:
Your right. 

Found a picture.
https://laptopkey.com/lenovo-thinkpad-multi-connect-bluetooth-keyboard-kt-1525-laptop-keyboard-keys/

Guess never saw this for sale in my area. My bad. 

Though if it's like the Thinkpad Keyboard 2 patch it wouldn't be much different:
https://patchwork.kernel.org/project/linux-input/list/?submitter=188019