lentinj/tp-compact-keyboard #49: disable lenovo_event_cptkbd

amosbird, 2020-02-13:
Hi, I'd like to let libinput to handle the middle button press instead of doing it inside the kernel. Is there a way to disable this feature in current hid-lenovo driver? Should I create a new driver to override it or is there better ways to disable this feature?

lentinj, 2020-02-13:
hid-lenovo isn't really handling middle-button behaviour either, it's built into the keyboard. None of the modes I know of on the keyboards produce unadultered trackpoint movements whilst the middle button is pressed, only scroll wheel events. Presumably this wasn't useful for them to add, given Windows doesn't have built-in scroll-emulation like Linux.

You could unbind the lenovo driver from the device and bind to the generic driver: https://lwn.net/Articles/143397/ or bind with your own driver. You'd then have to use ``--native-mouse-disable`` to return to the native middle button mode (an instantaneous press/release as you lift the middle mouse button).

amosbird, 2020-02-13:
> hid-lenovo isn't really handling middle-button behaviour either,

hmm, then what does this code do? https://github.com/torvalds/linux/blob/master/drivers/hid/hid-lenovo.c#L371

> You could unbind the lenovo driver from the device 

Does that mean I lose the ability to tune the trackpoint's sensitivity?

lentinj, 2020-02-13:
> > hid-lenovo isn't really handling middle-button behaviour either,
> 
> hmm, then what does this code do? https://github.com/torvalds/linux/blob/master/drivers/hid/hid-lenovo.c#L371

Suppress middle-click events if you middle-click-and-scroll. Otherwise you'll trigger paste (or whatever middle click does) every time you scroll.

amosbird, 2020-02-13:
> Suppress middle-click events

That's exactly what I want to disable :)  (override the lenovo_event_cptkbd function) And I don't think it's related to whether hid-lenovo handles middle-button behaviour.

amosbird, 2020-02-23:
> Suppress middle-click events if you middle-click-and-scroll. Otherwise you'll trigger paste (or whatever middle click does) every time you scroll.

btw, libinput also has this logic. I think we can get rid of it from kernel driver

lentinj, 2020-02-24:
> > Suppress middle-click events if you middle-click-and-scroll. Otherwise you'll trigger paste (or whatever middle click does) every time you scroll.
> 
> btw, libinput also has this logic. I think we can get rid of it from kernel driver

Yes, libinput is probably the more correct place for this sort of hackery nowadays. But I'm not sure there's any value in a painful transition process by removing it from the kernel. Implementing a sysfs configuration to disable it would be easy enough, and seems more preferable to me if this is something that's truly required.

OTOH, you may achieve very similar results with ``--native-mouse-disable`` in the userland tools, which off the top of my head will result in wheel events, then a middle button instantaneous down-up at the end. Due to the ordering it shouldn't trigger the middle-mouse-button suppression.

amosbird, 2020-02-24:
> Implementing a sysfs configuration to disable it would be easy enough

Could you give some hints over this route? ( if it's really easy enough...)

amosbird, 2020-02-24:
I just tried `--native-mouse-disable`. It doesn't pass the middle-mouse-button and the middle scroll is also disabled.

lentinj, 2020-03-02:
> > Implementing a sysfs configuration to disable it would be easy enough
> 
> Could you give some hints over this route? ( if it's really easy enough...)

I'd do something along these lines:

* Define ``middlebutton_state`` == 3 to disable the middle button / scroll disambiguation (i.e. update the comment at the top).
* Add code to ``lenovo_features_set_cptkbd()`` to set ``middlebutton_state`` = 0 or 3, depending on a new sysfs option you'll also need to add (just copy/paste how the fn_lock option is defined).
* In ``lenovo_event_cptkbd()``, just return 0 if ``middlebutton_state`` == 3 (you could just comment this out to see if this is the behaviour you really want)
* Remove ``cptkbd_data->middlebutton_state = 0;`` from ``lenovo_probe_cptkbd()``, as it will call ``lenovo_features_set_cptkbd()`` instead to set the value correctly.

amosbird, 2020-03-02:
Hmm, does that mean updating the driver and commiting to the kernel code base?

trevor403, 2020-03-09:
Yes but you could also build the driver out of tree and `insmod` it yourself. I've done this before.