lentinj/tp-compact-keyboard #51: ThinkPad keyboard II

dal00, 2020-04-30:
This looks interesting. Is anybody planning on getting it?
https://www.lenovo.com/us/en/accessories-and-monitors/new-arrivals/KBD-BO-TrackPoint-KBD-US-Eng/p/4Y40X49493

cblegare, 2020-09-16:
I got it with a fr-ca layout.  It's very nice.

I'm looking in this repository because Fn and FnLock do not seam to work out of the box.  I could post back when I find my way around that

dal00, 2020-09-16:
I think you should probably create another issue for the fn/fnlock thing

cblegare, 2020-09-16:
@dal00 

> I think you should probably create another issue for the fn/fnlock thing

I will do, thanks

> But are you able to compare it to the old one in quality and function?

I didn't try the old one, but I can compare it with the P50s, E560, Yoga and T500 I use daily.

The disposition is precise, keys feel less flacky when compared to an internal keyboard. It is thin and feels robust. Even though it is not a proper mecanical keyboard with actual mecanical switches, there is a feel of a "click", even if the sound itself is quieter than a internal keyboard.  

I found it was quite easy to pluck out by accident one the the legs that make the angle, which should not be an issue when handled with a least a bit of care.

The three buttons below the space bar are flat and low, not concave like they used to be.  The plastic finish looks a bit more textured and silky than it used to be, which is good in my opinion.

All in all, I am pleased with it!

cblegare, 2020-09-23:
Somewhat important thing to note: You can't register your product for this keyboard online, and must send a proof of purchase to get any warranty manually.

jeroney, 2020-10-03:
Works great out of the box. Though needs kernel level support for a few things.

It has 2 modes. Android & Windows. Really Windows is full keyboard mode. Where Android mode looks like it disables the FnLock functionality.

- FN control through sysfs
     - The keybaord now has a light to tell you that Fnlock is on or not. 
        It's controlled via the keyboard

- Some keys need fixing (through kernel or udev?)
          - Screen Brightness minus (F5) - not sending anything
          - Screen Brightens plus (F6) - not sending anything 
          - Project (sending ALT+P) (F7) - need to fix
          - Quote box? (F8) - not sending anything
          - Settings (F9) - not sending anything
          - Bluetooth (F10) - not sending anything
          - Keyboard (F11) - not sending anything
          - Favorite (F12) - not sending anything

Thinkpad II USB summary:
Vendor=0x17ef 
Product=0x60ee 

Thinkpad II Bluetooth summary:
Vendor=0x17ef 
Product=0x60e1

Thinkpad II Keyboard USB

lsusb -v
===============
Bus 003 Device 002: ID 17ef:60ee Lenovo TrackPoint Keyboard II
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x17ef Lenovo
  idProduct          0x60ee 
  bcdDevice            1.24
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
[... 219 lines cut ...]
B: KEY=3f000301ff 0 0 483ffff17aff32d bfd4444600000000 1 130ff38b17c000 677bfad9415fed 19ed68000004400 10000002
B: REL=1040
B: ABS=100000000
B: MSC=10

I: Bus=0005 Vendor=17ef Product=60e1 Version=0035
N: Name="TrackPoint Keyboard II System Control"
P: Phys=8c:c6:81:15:25:d8
S: Sysfs=/devices/virtual/misc/uhid/0005:17EF:60E1.0008/input/input36
U: Uniq=d3:0d:a0:d7:d1:8b
H: Handlers=kbd event26 
B: PROP=0
B: EV=13
B: KEY=c000 10000000000000 0
B: MSC=10

I: Bus=0005 Vendor=17ef Product=60e1 Version=0035
N: Name="TrackPoint Keyboard II"
P: Phys=8c:c6:81:15:25:d8
S: Sysfs=/devices/virtual/misc/uhid/0005:17EF:60E1.0008/input/input38
U: Uniq=d3:0d:a0:d7:d1:8b
H: Handlers=event27 
B: PROP=0
B: EV=9
B: ABS=10000000000

lentinj, 2020-10-04:
>     * Some keys need fixing (through kernel or udev?)
>       - Project (sending WINDOWS+P) (F7) - need to fix somehow?
>       - Quote box? (F8) - not sending anything
>       - Settings (F9) - not sending anything
>       - Bluetooth (F10) - not sending anything
>       - Keyboard (F11) - not sending anything
>       - Favorite (F12) - not sending anything

It would be interesting to use hcidump (for bluetooth) or hidraw (``cat /dev/hidraw1 | xxd -b -c6`` roughly) to see what these keys produce at that level. Although there's a high chance they all produce the same code until some special mode is enabled (this is what all the other keyboards I've tried do).

jeroney, 2020-10-04:
Pressing F8 & F11 don't produce anything.

[root@desktop jerone]# cat /dev/hidraw1 | xxd -b -c6
00000000: 00000101 11000001 00000101 00000000 00000101 10111100  ......
 <--  Quote box / Action Center (F8)
00000006: 00000101 00000000 00000101 10110110 00000101 00000000  ......
<--  Settings (F9)
0000000c: 00000101 10110111 00000101 00000000 00000101 10111000  ......
<-- Favorite (F12)

Also figured I did try updating the lenovo-hid driver in the kernel. That's
when I realized this thing is nothing like the previous model under the
hood.

So the keyboard has two modes. Android & WIndows. Windows it's a normal
keyboard. Android mode does two things:
 - Disables FNLock (so no F1 - F12)
- It Changes the function of the keys (F9 - F12) to what is displayed above
the key .. so the Bluetooth(F10) becomes the Home button.

So in Android mode everything is HID compliant. In Windows Mode it's not.

Really not sure why Lenovo keeps doing this. Just make HID compliant keys
and there wouldn't be a problem anywhere.

ValdikSS, 2021-10-17:
https://lore.kernel.org/linux-input/20211017083246.977096-1-iam@valdikss.org.ru/T/

This patch enables "native mode", in which middle button works properly (does not emulate press-release before scrolling) and horizontal scrolling is supported. It also handles trackpoint sensitivity and proper button mapping.

leoluk, 2021-10-19:
@ValdikSS Working horizontal scrolling? That's awesome! Do you need any help with testing?

jeroney, 2022-02-20:
Tested @ValdikSS patches posted on linux-input mailing list. Not sure what the hold up is. But works really well for both USB & Bluetooth modes of the keyboard.

I came up with udev rules to have fn-lock disabled. The bluetooth one definitely took some work to figure out. But old work actually came in handy and probably needs updating on the side for the original Thinkpad keyboard.

cat <<'EOF' > [/etc/udev/rules.d/99-disable-fn-lock-thinkpad-trackpoint-keyboard-II.rules](https://github.com/etc/udev/rules.d/99-disable-fn-lock-thinkpad-trackpoint-keyboard-II.rules)
#ThinkPad TrackPoint Keyboard II USB
SUBSYSTEM=="hid", \
    DRIVER=="lenovo", \
    ATTRS{idVendor}=="17ef", ATTRS{idProduct}=="60ee", \
    ATTR{fn_lock}="0"

#ThinkPad TrackPoint Keyboard II Bluetooth
SUBSYSTEM=="input", \
    ATTRS{id/vendor}=="17ef", ATTRS{id/product}=="60e1", \
    TEST=="/sys/$devpath/device/fn_lock", \
    RUN+="/bin/sh -c 'echo 0 > \"/sys/$devpath/device/fn_lock\"'"
EOF

jeroney, 2022-02-20:
Oh also latest version of the patches (trying) to make their way upstream:
         https://patchwork.kernel.org/project/linux-input/list/?submitter=188019

jeroney, 2022-05-02:
Good news. The patches are on their way to the Linux kernel in 5.19

Staging for the patches can be found here:
https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/log/?h=for-5.19/lenovo

bertogg, 2022-07-04:
Hi, I have the Bluetooth keyboard (17ef:60ee) and I'm trying to use the special keys. In my case:

  - Manage external display (F7) - works fine out of the box ✅
  - Action center (F8) - not sending anything ❌
  - Settings (F9) - works fine out of the box ✅
  - Bluetooth (F10) - works fine out of the box ✅
  - Keyboard (F11) - not sending anything ❌
  - Favorite (F12) - `acpi_listen` gives me `button/prog1`, and it's mapped to `XF86Launch1`, it doesn't do anything by default but I can configure it ✅

I'm using Linux 5.18.2 with [this patch](https://github.com/torvalds/linux/commit/24401f291dcc4f2c18b9e2f65763cbaadc7a1528) and udev 247.3.

ValdikSS, 2022-07-04:
@bertogg, are you on Xorg? Xorg has [8-bit key codes](https://gitlab.freedesktop.org/xorg/xserver/-/issues/260), while the KEY_NOTIFICATION_CENTER is 444 for example.

bertogg, 2022-07-04:
I'm with a GNOME Wayland session with Xwayland

bertogg, 2022-07-04:
Ah yeah, I see them with `libinput debug-events`. How can I use them in GNOME?

managervcf, 2022-07-10:
> Hi, I have the Bluetooth keyboard (17ef:60ee) and I'm trying to use the special keys. In my case:
> 
> * Manage external display (F7) - works fine out of the box white_check_mark
> * Action center (F8) - not sending anything x
> * Settings (F9) - works fine out of the box white_check_mark
> * Bluetooth (F10) - works fine out of the box white_check_mark
> * Keyboard (F11) - not sending anything x
> * Favorite (F12) - `acpi_listen` gives me `button/prog1`, and it's mapped to `XF86Launch1`, it doesn't do anything by default but I can configure it white_check_mark
> 
> I'm using Linux 5.18.2 with [this patch](https://github.com/torvalds/linux/commit/24401f291dcc4f2c18b9e2f65763cbaadc7a1528) and udev 247.3.

That's exactly what is happening with my keyboard as well. Also I found that the `FnLock` indicator always turns on on startup, even though I disable it before rebooting.

jeroney, 2022-07-10:
@managervcf 

To have FnLock disable you have to use udev to do it. I wrote a udev rule.

Run this command to install the udev rule, then restart udev or the system for it to take affect:

cat <<'EOF' > /etc/udev/rules.d/99-disable-fn-lock-thinkpad-trackpoint-keyboard-II.rules
#ThinkPad TrackPoint Keyboard II USB
SUBSYSTEM=="hid",
DRIVER=="lenovo",
ATTRS{idVendor}=="17ef", ATTRS{idProduct}=="60ee",
ATTR{fn_lock}="0"

#ThinkPad TrackPoint Keyboard II Bluetooth
SUBSYSTEM=="input",
ATTRS{id/vendor}=="17ef", ATTRS{id/product}=="60e1",
TEST=="/sys/$devpath/device/fn_lock",
RUN+="/bin/sh -c 'echo 0 > "/sys/$devpath/device/fn_lock"'"
EOF