lentinj/tp-compact-keyboard #52: Enabling FnLock with ThinkPad TrackPoint Keyboard II

cblegare, 2020-09-16:
I'm looking in this repository because Fn and FnLock do not seam to work out of the box. I could post back when I find my way around that.

The Esc key light won't get green, even using the Fn key (along with the Mic key to get an F4) do not work.  I like my F* keys handy and media keys available with the Fn key.

As per the readme, I tried to control whether fn_lock is enabled by issuing:

```shell
echo 0 > /sys/bus/hid/devices/*17EF\:604*/fn_lock 
```

and no folder matched this pattern. On my laptop, the `/sys/bus/hid/devices/` folder contains

- 0003:046D:4016.0005
- 0003:046D:401B.0004
- 0003:046D:C52B.0001
- 0003:046D:C52B.0002
- 0003:046D:C52B.0003
- 0003:17EF:60EE.0009
- 0003:17EF:60EE.000A
- 0003:17EF:60EE.000B

And I am out of my confort zone here. any advice?

lentinj, 2020-09-22:
The tools / kernel drivers developed here haven't been tested with the ThinkPad TrackPoint Keyboard II, thus why it doesn't match (``60EE`` vs. ``604*``).

I don't know much about the new keyboards, and don't own one, so no idea what's needed to enable Fn-lock. You could try using/modifying the tool here: https://github.com/lentinj/tp-compact-keyboard/tree/master/tp-compact-usb-keyboard and see if it's able to change settings of the keyboard. However, seems pretty unlikely that the commands to change keyboard mode would be exactly the same I'm afraid.

cblegare, 2020-09-23:
Looks like I might have an issue with the keyboard itself.  I have a support case and should receive a new one.  I'll post back then

dlipovetsky, 2020-12-09:
When I move the "windows/android" switch on the back of the keyboard to "windows," the function lock works.

However, the keycodes are not mapped to function keys. Here are the keycodes of F1 to F12:
```
showkey --keycodes
```
```
^[OPkeycode  59 press
keycode  59 release
^[OQkeycode  60 press
keycode  60 release
^[ORkeycode  61 press
keycode  61 release
^[OSkeycode  62 press
keycode  62 release
^[[15~keycode  63 press
keycode  63 release
^[[17~keycode  64 press
keycode  64 release
^[[18~keycode  65 press
keycode  65 release
^[[19~keycode  66 press
keycode  66 release
^[[20~keycode  67 press
keycode  67 release
^[[21~keycode  68 press
keycode  68 release
^[[23~keycode  87 press
keycode  87 release
^[[24~keycode  88 press
keycode  88 release
```

cblegare, 2023-02-25:
> When I move the "windows/android" switch on the back of the keyboard to "windows," the function lock works.

Oh, that might be it. I think we can close this issue

closed 2023-02-25 by cblegare