lentinj/tp-compact-keyboard #54: thinkpad compact

developer91234, 2021-04-26:


lentinj, 2021-04-27:
No, in short. I'm pretty sure others have already forced these scripts to talk to a compact II and it's not responded. You'd have to follow a similar process of sniffing the HID packets going to the keyboard under windows to get any further.

I've not got one of the keyboards myself, the original BT compact keyboard is still going strong, and not sure I will after the glowing review :)

developer91234, 2021-04-27:
> I've not got one of the keyboards myself, the original BT compact keyboard is still going strong, and not sure I will after the glowing review :)

Glad I saved you the trouble :)

I think vertical scrolling behavior may be fixable to stop it from pasting after every scroll event, but with horizonal scroll no event is even emitted so I don't know where to begin.

lentinj, 2021-04-28:
> vertical scrolling behavior may be fixable to stop it from pasting after every scroll event

[hid-lenovo has logic to do manage this for the compact keyboards](https://github.com/torvalds/linux/blob/87fcb6a69e54527ccaeb5878168cccdba9f6b6ae/drivers/hid/hid-lenovo.c#L357-L390), but there may be an easier option to manage this in libinput than slogging through updating the kernel module.

> horizonal scroll no event is even emitted so I don't know where to begin.

IIRC this keyboard has a bunch of compatibility modes (Android, Windows, Mac). It'd be worth trying under each with the proper drivers to see if it's at least possible. You may find one mode is less buggy than another.

developer91234, 2021-04-28:
I tried both modes and it's the same. There's no mac mode - nobody uses lenovo products with mac.

Maybe I need to add a special .conf file to `/etc/X11/xorg.conf.d/`for this crap? Maybe turn on wheel emulation explicity or even downgrade the driver to synaptic? Seems like it's using libinput right now but I'm not sure how to tell if it's libinput or evdev. To reiterate, the problem is that middle click pastes after you touch the trackpoint to scroll. This is on Manjaro 5.11. With the very latest Ubuntu it pastes right away on middle down. Same kernel, I can't explain the difference. But the behavior is broken on both.

I think the first thing I'll try is a bleeding edge kernel.

developer91234, 2021-04-29:
OK I can't get it working short of disabling middle paste. And I can't get horizontal to work obviously.

If you want to try rescuing this device for linux systems I'll be happy to test whatever you come up with, let me know.

ValdikSS, 2021-10-17:
https://lore.kernel.org/linux-input/20211017083246.977096-1-iam@valdikss.org.ru/T/

This patch enables "native mode", in which middle button works properly (does not emulate press-release before scrolling) and horizontal scrolling is supported.