lentinj/tp-compact-keyboard #59: fn-lock-enable  fail

githublaohu, 2023-08-13:
OS：Ubuntu22.04

keyboard:  TrackPoint Keyboard II  and  ThinkPad Compact Bluetooth Keyboard with TrackPoint

Scheme 1 

```
open@open:/source/tp-compact-keyboard/tp-compact-bt-keyboard$ sudo ./tp-compact-keyboard --fn-lock-disable
ThinkPad Compact Bluetooth Keyboard with TrackPoint No keyboard found!
hidraw2
Disabling Fn lock for /dev/hidraw2...
open@open:/lsource/tp-compact-keyboard/tp-compact-bt-keyboard$ sudo ./tp-compact-keyboard --fn-lock-enable
ThinkPad Compact Bluetooth Keyboard with TrackPoint No keyboard found!
hidraw2
Enabling Fn lock for /dev/hidraw2..

```
Scheme 2
```
root@open:/source/tp-compact-keyboard/tp-compact-bt-keyboard# echo 0 > /sys/bus/hid/devices/0005\:17EF\:60E1.0007/fn_lock
root@open:/source/tp-compact-keyboard/tp-compact-bt-keyboard# cat /sys/bus/hid/devices/0005\:17EF\:60E1.0007/fn_lock
0

```

jeroney, 2023-08-13:
This will not work in Ubuntu 22.04 as it is running kernel 5.15 and Ubuntu
did not backport the thinkpad keyboard patches to it.

The patches came in kernel 5.19. So you would need to use a newer version.

You can file a support case with Ubuntu to see if they will backport the
patches to 5.15.

githublaohu, 2023-08-13:
> 

ubuntu kenrnel  
```
Linux open 6.2.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
````

jeroney, 2023-08-13:
Ah. Oh not sure about the command tp-compact-keyboard .. but it's not able
to see the keyboard properly when connected via bluetooth is likely.

If you look at the udev rules here under "disable fn-lock by default":
https://github.com/lentinj/tp-compact-keyboard

You can see it has to do a trick for bluetooth connections that it doesn't
have to do for usb connection.

Would say it's best to use the udev rule and then enable/disable manually.

jeroney, 2023-08-16:
> How can I manually disable it
>
>         echo Enabling native mode for F7/F9/F11 for /dev/$HIDRAW...
>         /bin/echo -ne "\x18\x01\x03" | dd of=/dev/$HIDRAW 2>/dev/null
>
> Is that so?

For thinkpad 2 keyboards. You'll need to be root:
           echo 0 > /sys/bus/hid/devices/*17EF\:60e*/fn_lock

For thinkpad 1 keyboards. You'll need to be root:
            echo 0 > /sys/bus/hid/devices/*17EF\:604*/fn_lock

The udev rule is best. It will disable it on plug-in. Then you can control
via keyboard.

githublaohu, 2023-08-17:
why
`
root@open:/lsource/tp-compact-keyboard/tp-compact-bt-keyboard# echo 1 > /sys/bus/hid/devices/0005\:17EF\:60E1.0020/fn_lock 
root@open:/lsource/tp-compact-keyboard/tp-compact-bt-keyboard# /bin/echo -ne "\x18\x05\x01" | dd of=/dev/hidraw2
记录了0+1 的读入
记录了0+1 的写出
3字节已复制，2.9186e-05 s，103 kB/s

`

githublaohu, 2023-08-17:
How can I check if the driver is installed

githublaohu, 2023-08-18:
Hello, I have three generations of TP compact keyboards, and the first generation settings are effective. 

```
root@open:/sys/bus/hid/devices/0005:17EF:6048.0022# echo 1 > fn_lock
root@open:/sys/bus/hid/devices/0005:17EF:6048.0022# cat fn_lock 
1
```
Both the second and third generations failed

githublaohu, 2023-08-18:
May I ask where the udev rules for TrackPoint Keyboard II can be found